LilyGO/TTGO-TAudio #2: bt_speaker does not compile - error: implicit declaration of function 'esp_a2d_register_data_callback'

N0rbert, 2018-04-14:
What I did:

1. installed ESP-IDF into ~/esp as recommended in the [official documentation](http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup.html) and then successfully compiled [hello_world project](http://esp-idf.readthedocs.io/en/latest/get-started/index.html#start-a-project);
1. cloned this repository with `git clone https://github.com/LilyGO/TTGO-T9-RGB_LED-WM8978.git` (latest commit ac508f0191031794072ed5a39dfc69080d6fbd03);
1. tried to compile *bt_speaker* with the following actions:

```
cd TTGO-T9-RGB_LED-WM8978/bt_speaker
make # used defaults anywhere
```

Got error message:

> TTGO-T9-RGB_LED-WM8978/bt_speaker/components/bt_speaker/./bt_speaker.c:90:9: error: implicit declaration of function 'esp_a2d_register_data_callback' [-Werror=implicit-function-declaration]
>          esp_a2d_register_data_callback(bt_app_a2d_data_cb);
>          ^
> cc1: some warnings being treated as errors
> /home/user/esp/esp-idf/make/component_wrapper.mk:273: recipe for target 'bt_speaker.o' failed
> make[1]: *** [bt_speaker.o] Error 1
> /home/user/esp/esp-idf/make/project.mk:457: recipe for target 'component-bt_speaker-build' failed
> 

Note: I'm using ESP-IDF on fully updated 64-bit Ubuntu 16.04 LTS.

cjdell, 2018-09-18:
Changing the function reference from:
`esp_a2d_register_data_callback`
to:
`esp_a2d_sink_register_data_callback`
seemed to fix it for me.

xuhongv, 2019-06-05:
good jod