LilyGO/TTGO-TAudio #7: Battery

CanMark, 2019-06-22:
Hi, 
I wonder do you know what battery should be connected to this board.
I have tried a 3.7v lipo but it does not appear to work.
Thanks,
Mark

vvs551, 2019-07-09:
Battery is working but IP5306 key is off (
You can try to connect pin_5 of ip5306 connect to GND and battery power will start
But, unfortunately it will stop again in 1 minute ((

CanMark, 2019-07-10:
So it should work off a 3.7v battery?

Are you suggesting that I connect the battery to a couple of pins on one of the chips on the board?

who4you, 2019-07-28:
I have the same problem with the battery always turns off after 35 seconds. (Programming with arduino on platform.io). Does anyone have a glue?

renebohne, 2020-05-18:
In order to solve this issue for my board, I connected (soldered wires) pin IO23 (of ESP32) to pin LED2 (of IP5306)  and pin IO5 to pin LED1.

My IP5306 supports I2C, so I was able to add this code in my setup() or main() methods:
bool ok = setPowerBoostKeepOn(true);

and this is the function that I took from the M5stack project:
bool setPowerBoostKeepOn(bool en){
    uint8_t data;
    Wire.beginTransmission(IP5306_ADDR);
    Wire.write(IP5306_REG_SYS_CTL0);
    Wire.endTransmission();

    if(Wire.requestFrom(IP5306_ADDR, 1))
    {
        data = Wire.read();

        Wire.beginTransmission(IP5306_ADDR);
        Wire.write(IP5306_REG_SYS_CTL0);
        if (en) Wire.write(data |  BOOST_OUT_BIT); 
        else    Wire.write(data &(~BOOST_OUT_BIT));  
        Wire.endTransmission();
        return true;
    }
    return false;
}