LilyGO/TTGO-TAudio #12: Example Code - WM8978 Arduino playing Icecast streams.

NonaSuomy, 2020-12-07:
This is a quick example for people like me that were looking for audio sample code.

**Debug Settings**

To see the serial output in the Arduino IDE enable the logging level to Info or higher. 
Tools -> Core Debug Level -> Info.
Now open the Serial Monitor. 

**Other settings**

Board: ESP32 DEV Module
Upload Speed: 921600
CPU Frequency: 240MHz (WiFi/BT)
Flash Frequency: 80MHz
Flash Mode: DIO
Flash Size: 4MB (32Mb)
Partition Scheme: Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)
Core Debug Level: Info
PSRAM: Enabled
Port: Pick your devices serial port that shows up.

**Code**

```
// Required Libraries (Download zips and add to the Arduino IDE library).
#include <WM8978.h> // https://github.com/CelliesProjects/wm8978-esp32
#include <Audio.h>  // https://github.com/schreibfaul1/ESP32-audioI2S

// T-Audio 1.6 WM8978 I2C pins.
#define I2C_SDA     19
#define I2C_SCL      18

// T-Audio 1.6 WM8978 I2S pins.
#define I2S_BCK     33
#define I2S_WS      25
#define I2S_DOUT  26
#define I2S_DIN     27

// T-Audio 1.6 WM8978 MCLK gpio number?
#define I2S_MCLKPIN  0

Audio audio(I2S_BCK, I2S_WS, I2S_DOUT);
WM8978 dac;

void setup() {
  // Setup wm8978 I2C interface.
  if (!dac.begin(I2C_SDA, I2C_SCL)) {
    ESP_LOGE(TAG, "Error setting up dac: System halted.");
    while (1) delay(100);
  }

  // Select I2S MCLK pin
  audio.i2s_mclk_pin_select(I2S_MCLKPIN);

  // WiFi Settings here.
  WiFi.begin("EnterSSIDHere", "EnterPasswordHere");
  while (!WiFi.isConnected()) {
    delay(10);
  }
  ESP_LOGI(TAG, "Connected. Starting MP3...");
  // Enter your Icecast station URL here.
  audio.connecttohost("http://hestia2.cdnstream.com/1458_128"); 
  // Volume control.
  dac.setSPKvol(40); // Change volume here for board speaker output (Max 63).
  dac.setHPvol(52, 52); // Change volume here for headphone jack left, right channel.
}

void loop() {
  // Start the stream.
  audio.loop();
}
```

arifshanji, 2021-08-06:
Thanks @NonaSuomy

Updated example because the `ESP32-audioI2S` library has been updated.

`// Required Libraries (Download zips and add to the Arduino IDE library).
#include <WM8978.h> // https://github.com/CelliesProjects/wm8978-esp32
#include <Audio.h>  // https://github.com/schreibfaul1/ESP32-audioI2S

// T-Audio 1.6 WM8978 I2C pins.
#define I2C_SDA     19
#define I2C_SCL     18

// T-Audio 1.6 WM8978 I2S pins.
#define I2S_BCK     33
#define I2S_WS      25
#define I2S_DOUT    26
#define I2S_DIN     27

// T-Audio 1.6 WM8978 MCLK gpio number?
#define I2S_MCLKPIN  0

Audio audio;
WM8978 dac;

void setup() {

  // Setup wm8978 I2C interface.
  if (!dac.begin(I2C_SDA, I2C_SCL)) {
    ESP_LOGE(TAG, "Error setting up dac: System halted.");
    while (1) delay(100);
  }

  // Select I2S pins
  audio.setPinout(I2S_BCK, I2S_WS, I2S_DOUT);
  audio.i2s_mclk_pin_select(I2S_MCLKPIN);

  // WiFi Settings here.
  WiFi.begin("EnterSSIDHere", "EnterPasswordHere");
  while (!WiFi.isConnected()) {
    delay(10);
  }
  ESP_LOGI(TAG, "Connected. Starting MP3...");
  // Enter your Icecast station URL here.
  audio.connecttohost("http://hestia2.cdnstream.com/1458_128"); 
  // Volume control.
  dac.setSPKvol(40); // Change volume here for board speaker output (Max 63).
  dac.setHPvol(52, 52); // Change volume here for headphone jack left, right channel.
}

void loop() {
  // Start the stream.
  audio.loop();
}`

NonaSuomy, 2021-08-08:
You're welcome and thanks for the update. If you build something cool let me know.