LilyGO/TTGO-TAudio #13: Microphone Example

haaslukas, 2021-03-05:
Is there any example available to get the microphone running? Or did someone succeed in doing so?

In my case, `i2s_read` always returns zero: `(i2s_read):rx NULL`

Here's my demo code:

```
#include <WM8978.h> /* https://github.com/CelliesProjects/wm8978-esp32 */
#include <Audio.h>  /* https://github.com/schreibfaul1/ESP32-audioI2S */

/* TTGO T-Audio I2C pins */
#define I2C_SDA     19
#define I2C_SCL     18

/* TTGO T-Audio pins */
#define I2S_BCK     33
#define I2S_WS      25
#define I2S_DOUT    26
#define I2S_DIN     27

/* TTGO T-Audio WM8978 MCLK gpio number */
#define I2S_MCLKPIN  0

Audio audio;
WM8978 dac;

void setup() {

  Serial.begin(115200);
  
  /* Setup wm8978 I2C interface */
  if (!dac.begin(I2C_SDA, I2C_SCL, 44100)) { // 44100 Sampling Rate
    Serial.println("Error setting up dac. System halted");
    while (1) delay(100);
  }
  
  dac.cfgI2S(2, 0); //Default: 2, 0 --> Philips 16bit
  dac.cfgADDA(1, 1); //Enable ADC DAC
  dac.cfgInput(1, 0, 0); //mic, linein, aux - Default: 0, 0, 0
  dac.cfgOutput(1, 0); // Output enabled, bypass disabled
  dac.setMICgain(25); // Default: 0
  dac.setAUXgain(0); 
  dac.setLINEINgain(0);
  dac.setSPKvol(40); // Dafault: 0
  dac.setHPvol(50, 50); //Default: 0, 0

  /* set i2s pins */
  audio.setPinout(I2S_BCK, I2S_WS, I2S_DOUT, I2S_DIN);

  /* Start MCLK */
  audio.i2s_mclk_pin_select(I2S_MCLKPIN);
 
}

void loop() {

  int32_t sample;
  size_t bytes_read;

  i2s_read(I2S_NUM_0, &sample, 4, &bytes_read, 2); 

  Serial.printf("%d,", sample);
  Serial.println(bytes_read);
}
```

NonaSuomy, 2023-05-10:
Did you ever figure it out?

NonaSuomy, 2023-05-11:
[TAudio Libraries.zip](https://github.com/LilyGO/TTGO-TAudio/files/11453478/TAudio.Libraries.zip)
There is a recording example from dfrobots in the wav folder here.